FRC-Team-620/Warbots2023 #73: Constants Housekeeping

m10653, 2023-02-08:
Clean up constants file and drive ports. Remove unused constants and move robot specific constants into their own drive ports.

transorsmth, 2023-02-09:
Is there any way to simplify driveports? I dislike needing to add it to every file, and needing a method, and not just having a default.

m10653, 2023-02-09:
There are a few way. The current way the driveports are setup is kind of nice because it forces one to think about the configuration of all the other robots so you dont have to worry about one changing breaking all of the others. The downside is that it requires a bunch more boiler plate.

One alternative would be to have a robotdriveports class the implements the idriveports interface. Then for other robots extend the robotdriveports and overwrite the required methods that needed to be changed. Downside to this is it would violate some of the assumptions for OOP inheritance and could have issues where updates to the base class are not anticipated by the extended classes.